TorstenRobitzki/bluetoe #22: Move to a more mature TMP library

TorstenRobitzki, 2018-05-16:
Especially with GCC, compile times are rising fast, when having real-world projects and using notification priorities.

http://metaben.ch seems to be a good list to pick one library from.

kaofishy, 2018-06-22:
I've now been bitten by this: apparently all the template metaprogramming is too much for Visual Studio to handle; when I tried to instantiate a `cstring_value<>` the entire IDE froze and had to be restarted. This has happened without fail, and my "solution" so far has been to switch to Eclipse.

TorstenRobitzki, 2018-06-22:
Well, I guess that Visual Studio would freeze even more when I switch Bluetoe to one of the libraries mentioned at metaben.ch.

kaofishy, 2018-06-22:
Hats off to whoever have to write live parsers for C++ templates tbqh.