TorstenRobitzki/bluetoe #37: change CI

TorstenRobitzki, 2018-11-02:
Travis CI seems to have a fixed maximum runtime of 10 minutes. Running make with -j results in build failures during the compilation of the tests (out of memory).

A new CI should:
- enable us to build the tests with several compilers and versions.
- build the documentation and store them on github.
- let us install cross compilers (and required vendor-libraries) to build examples.
- let us save the example binaries for beeing tested (minor).
- should be affordable by our sponsor (~10€/Month).

If we host a CI server on our own, we could add additional hardware to do integration tests with different hardware bindings (but maybe this shouldn't be scope of this issue).

TorstenRobitzki, 2018-11-07:
Assumed, we would set up a Jenkins server on our own: We could run HIL tests (based on the Scheduled Radio Interface) for nrf51 and nrf52.

TorstenRobitzki, 2018-11-17:
@obruns Do you have something to contribute?

TorstenRobitzki, 2018-11-19:
I can't merge the pullrequest on Gitlab (most likely, because the repo is hosted on Github). But if I got it right, it's just this two files (.gitlab-ci.yml and Dockerfile). I will just "merge" them by hand into master here...

TorstenRobitzki, 2018-11-19:
I just realized that the pull request was on your branch. I guess to fully use Gitlab as CI, we would have to move this project to Gitlab.

obruns, 2018-11-19:
Mmh, sure about the move to Gitlab. That's what I implied when we talked on Friday.

My reply from earlier today was just to give you a glimpse of what I tried to do with Gitlab CI so far.

TorstenRobitzki, 2018-11-19:
I would like to go with the following approach:
- Use a cheap virtual server somewhere (~5€/month)
- create a docker image to run jenkins 
- create some docker images with different compilers (or just one as @obruns did)
- Let the Jenkins server report the build state to Github
- Let the Jenkins server do nightly build to collect interesting stuff like image sizes, pre-build binaries, documentation and such. (guess that means we would have to have a data base then).

TorstenRobitzki, 2018-11-19:
Here are some numbers from the Gitlab builds: https://gitlab.com/TorstenRobitzki/bluetoe/pipelines/37166668/builds

TorstenRobitzki, 2026-09-21:
The last changes to the CI implemented already a lot of this. What remains is basically, the documentation and example binaries that could be used for testing the library without the need to compile. 

Maybe the release feature of GitHub could be used to provide zip files every once in a while that contains the library, the documentation and examples.