TorstenRobitzki/bluetoe #43: Compiling a simple example results in static assertion

PeterTaps, 2019-04-11:
I just built bluetoe code on my Ubuntu 18.04 box. The build went fine.

Next, I copied simple_server.cpp to mysvr.cpp in a new directory.

Here is my simple command to build it:
g++ \
   -I$ThirdParty/bluetoe \
   -I$ThirdParty/bluetoe/bluetoe/utility/include \
  mysvr.cpp -o mysvr

I still need to link with libraries but that will come once I overcome my problem.

When I try to compile, I am getting the following assertion from server.hpp:

    static_assert( number_of_client_configs != 0, "there is no characteristic that is configured for notification or indication" );

Is it something that I am doing wrong?

Thanks
Peter

TorstenRobitzki, 2019-04-12:
Hi Peter,
the simple_server.cpp is an example, that is not meant to run anywhere. Could you please base your example on one of the other examples (blinky.cpp for example).

No, you did nothing wrong, the simple_server.cpp example is simply not working anymore. If you look at the bottom of the CMakeList.txt file in the example folder, you will see all examples that a regularly build and should work out of the box.

I will keep this issue open, as a reminder, that the simple_server examples is broken and needs to be either removed or fixed. Thanks for reporting!

regards,
Torsten

PeterTaps, 2019-04-12:
Thanks Torsten. I need to run the code as a BLE server on Ubuntu. I removed all the nrf-specific logic from blinky.cpp except the following line:

device<blinky_server> gatt_srv;

Here, the definition for device is coming from nrf51-specific device.hpp. 

How do I create a ubuntu-specific generic device?

Appreciate your help.

Regards,
Peter

TorstenRobitzki, 2019-04-13:
Hi Peter,
currently, Bluetoe runs just on Nordic Microcontrollers. So currently, it is not able to run a GATT server on Ubuntu.

regards,
Torstem

PeterTaps, 2019-04-15:
Thanks Torstem. I tried to create a plain Ubuntu version by removing Nordic stuff out of nrf51.hpp. However, it turned out to be quite some work.

Do you have any plans to extend this library for Raspberry Pi? Thanks.

TorstenRobitzki, 2019-04-15:
Yes, there are plans to implement HCI (https://github.com/TorstenRobitzki/bluetoe/issues/29), which then should be / could be used to use vanilla PC hardware.

TorstenRobitzki, 2021-02-10:
I don't see any further actions.

closed 2021-02-10 by TorstenRobitzki