TorstenRobitzki/bluetoe #102: Connection lost when receiving twice the same packet from the Central

XavierBoniface, 2023-01-13:
Occurs when the Master misses the ACK from the Peripheral and resends the same packet.
Observations: the Peripheral re-sends the same ACK. Then it starts misbehaving: it opens Rx windows on subsequent connection events but does not receive the packets sent by the Central (Rx windows time out and get wider and wider). 
[image]

Could probably be reproduced by lowering the Tx power of the Peripheral after some time in the connection, in order to cause the Central to miss the Ack and re-send the same packet.

TorstenRobitzki, 2023-01-25:
Maybe this is related or was even introduced with the fix to  #70?!

xnlcasad, 2023-01-25:
It does sound related !

TorstenRobitzki, 2023-01-27:
Seems to be the same bug. Classification of the received PDU was factored out to `radio_hardware_xxx::received_pdu()` and this function can only return 3 different situations: "PDU received correctly", "timeout" or "PDU not received ok but anchor valid".

For the implementation of encryption there seems to be the need for the detection of the situation of "CRC ok, but MIC not ok". Which then is an indication, that a PDU was resend by the master. To be totally clear, one could try to decrypt the PDU with a decremented counter. But probably wouldn't learn a lot by doing so. For sure, it's important, that a PDU without CRC error, but MIC error is used to acknowledge a message, but is itself not queued into the input queue.

commit 8ae41c52 (2023-01-30): add acknowledge() to ll_data_pdu_buffer in preparation to fix #102
commit 861b5ad2 (2023-01-30): fix #102 by acknowlageing a resend PDU with valid CRC but invalid MIC